Infineon/TLE5012-Magnetic-Angle-Sensor #1: Example connection scheme for tle5012be1000 chip and microcontorller

alexxy, 2018-11-18:
Hi!

Is there a simple connection scheme between encoder chip and uC?

MorganSandercock, 2019-02-05:
Yes...
connect CSQ(3) to any convenient uC pin to use as chip-select
connect DATA(4) to MISO
connect SCK(2) to SCK
connect DATA(4) to MOSI via a resistor. Use anything between 3.3K and 10K.

If you don't use a resistor between DATA and MOSI then it is possible to modify the library to put MOSI into a high-impedance state after writing the command word and before reading the response. Setting it back to SPI output mode depends on the specific microcontroller, so this is not a simple one-line change to the library.

OlafFilies, 2019-08-20:
Table with the connection schema is now included.

closed 2019-08-20 by OlafFilies

MinSoeVertex, 2019-12-21:
Should we use  SSC interface (push-pull configuration) to connect with Arduino in SPI mode? If so, according to diagram, SCK and CSQ should use 100ohm and DATA should use 470ohm. Can you clarify that please?

OlafFilies, 2020-01-09:
The SSC interface is the standard interface working on all versions of the TLE5012B chip. Beside that each version has a second default interface depending on what you have ordered as IIF, PWM or SPC which you can directly use. Please have a look in [here](https://www.infineon.com/dgdl/Infineon-Angle_Sensor_TLE5012B-UM-v01_02-en-UM-v01_02-EN.pdf?fileId=5546d46146d18cb40146ec2eeae4633b) for the right connection of these.
As shown in Figure 3.7 of this manual the 100ohm for SCK and CSQ, as well as the 470ohm for the DATA. Keep in mind, that SSC is a 3-wire SPI, which means that you have to put a  >3.3kohm between MISO and MOSI if you use a bulk chip (see table in Readme).

trotamares, 2020-02-18:
Hi. The library fails to compile for ESP8266: C:\Users\pc\Documents\Arduino\libraries\TLE5012B\src\Tle5012b.cpp: In constructor 'Tle5012b::Tle5012b()':

C:\Users\pc\Documents\Arduino\libraries\TLE5012B\src\Tle5012b.cpp:147:17: error: invalid conversion from 'SPIClass*' to 'Tle5012b_SPI*' [-fpermissive]  _spiConnection = &SPI; 

Can anybody help me? Thank you

OlafFilies, 2020-02-18:
I assume that you have a bulk TLE5012B chip and if you want to use it with an ESP than you should use the Tle5012b_4wire library instead and connect the chip for the SSC interface. An alternative can be, but that needs more investigation, by inserting an ESP switch into the  util/Tle5012b_SPI library.
Anyway, I will have a look at it but this will take some time.