Infineon/TLE5012-Magnetic-Angle-Sensor #6: Reset sensor to 0 position

kristsm, 2019-04-24:
Hi! Is there a command to reset TLE5012 to zero position (0 angle value and 0 revolutions count) after arduino has started? 
Otherwise, after restart , the sensor 0 position is always on where it's left.  It's also hard to understand from the datasheet if it's even possible?

Aeumel, 2019-04-24:
Hello, you can reset the sensor of course, therefore the register ACSTAT has top be filled with 0x401, which will trigger the AS_FRST (Activation of Firmware reset) and AS_RST (Activation of Hardware reset). Have a look at this function (will be part of the next release of this library).
```
errorTypes Tle5012b::resetFirmware()
{
	uint16_t rawData = 0x401;
	errorTypes status = writeActivationStatus(rawData);
	return (status);
}
```
Call this and the triggerUpdate() function in your sketch. Angle and speed will be set to 0, but right the next readout will set the angle value back to the relative position of the magnet to the sensor. So it will always show the actual position of the magnetic field and therefore of the motor or shaft or whatever you measure with it. Reset firmware will reset whatever you have change in the sensor register, like offsets, slave number or interface settings.

KOMsandFriends, 2019-06-22:
Hi @kristsm, has that solved your problem?

closed 2021-07-12 by OlafFilies