Infineon/TLE5012-Magnetic-Angle-Sensor #9: SPI speed and delay

kristsm, 2019-04-25:
The datasheet advertises 8mbit SPI speed. With your library default settings getanglevalue takes almost 300us to read actual angle, which is quiet slow for any fast enough positioning device. I tried with speed= 8000000 and I get 0 value readings. Most I could get with my uC ( it supports 80mhz SPI easily) speed = 2500000. Together withcmodified DelayuS of 5uS I get about 50us getangle read time. Why so? And why is this delay function there? Can the reading be faster than that?

Aeumel, 2019-04-25:
Hello, yet there are too many resets and delays in the library witch are not necessary. If you use the Sensor2Go version of this sensor (the one with XMC2Go attached) than you need a 5µs delay between the data send and receive part for switching. All other delays are not needed except a 5µs delay in the trigggerUpdate function (Yet there is 1ms). Also remove this from the readFromSensor function:

```
	if (command & CHECK_CMD_UPDATE)
	{
		triggerUpdate();
	}
```

This is the wrong place for the triggerUpdate. With an XMC2Go attached the maximum baudrate can be 1000000 baud, which is the maximum XMC1100 allows. BTW, I will ask my colleagues how fast this sensor can be and I'm pretty sure that it can be much faster but this will need Infineons Dave IDE, at least yet.

kristsm, 2019-04-29:
Aeumel, thanks for your response! I tried to remove this triggerupdate function from readFromSensor function, but in my case it even adds few us to overall read speed (strange). With SPI speed set to 2mbit and DelayuS to 1us  I get about 40us read speed. 

Still, there is some wall I hit when I set speed to more than 2mbit - even 2.1mbit or 2.01mbit is not possible. I think there is some software limit somewhere I cannot understand. If I read datasheet, it says in push-pull mode , the SSC interface max baud rate is 8mbit, but with open-drain - 2mbit. This is a very strange coincidendce, right? :) My sensor is E1000 series and by default it should be set to push-pull SSC configuration.  According to TLE5012 Register settings document, Mod_3 SSC_OD sets the sensor to open-drain or push-pull configuration. I tried to read this register setting and if I am right, that SSC_OD bit is set to "0", which is right for high-speed push-pull setting. But still there is a 2mbit speed wall in my case. Perhaps there is something set in this Infineon library set wrong to limit the user to go more than 2mbit in push-pull configuration?

My hardware is set in push-pull configuration as follows:
1. IFA/IFB/IFC are left open;
2. CSQ and SCK has 100ohm resistor;
3. DATA line has 510ohm resistor (could not find the suggested 470ohm in my shelf);
4. DATA(4) connected to MOSI via 3.3K resistor.

I hope for someone from Infineon to comment on my problem!

Krists

kristsm, 2019-04-29:
>  I will ask my colleagues how fast this sensor can be and I'm pretty sure that it can be much faster but this will need Infineons Dave IDE, at least yet.

Still it will never be faster than 42.7uS Tupd rate from the datasheet , right?

Aeumel, 2019-04-29:
Hello, please have a look at the user manual of the TLE5012B ([https://www.infineon.com/cms/en/product/sensor/magnetic-sensors/magnetic-position-sensors/angle-sensors/tle5012b-e1000/](https://www.infineon.com/cms/en/product/sensor/magnetic-sensors/magnetic-position-sensors/angle-sensors/tle5012b-e1000/). This will give you the complete information about this sensor. Specially have look at chapter 5. You see a 2Mbit/s boundary for the fast SSC Interface, which is true for the open drain configuration. If you want more than you have setup a push-pull configuration on the SSC Interface which goes up to 8Mbit/s. For fast read out you have switch off also the safety word. Also keep in mind that even if you the E1000 version with a predefined IIF/SSC interface, it is possible to reset the sensor to any almost any other interface (see chapter 3) specially 3.7 and 3.8. So if you have a 3.3K resistor to MOSI, than you have set it to the SSC open-drain bus system. You may have to readout several registers to really identify the sensors settings. An yes it will not be faster than 42.7us when set in FIR_MD register. BTW, if this is really not fast enough, than you may have a look at the new [TLE5501](https://www.infineon.com/cms/en/product/sensor/magnetic-sensors/magnetic-position-sensors/angle-sensors/tle5501-e0001/) which has a bandwidth of 1E6deg/s.

OlafFilies, 2021-07-12:
We will address the speed problem with the next version of the library.

closed 2021-07-12 by OlafFilies

dzid26, 2023-04-25:
> > I will ask my colleagues how fast this sensor can be and I'm pretty sure that it can be much faster but this will need Infineons Dave IDE, at least yet.
> 
> Still it will never be faster than 42.7uS Tupd rate from the datasheet , right?

Details regarding the delay:
[image]

In any case, regardless the delay it can be compensated in the software like so:
https://github.com/dzid26/StepperServoCAN/blob/f7aa59a2d4ad878b0b4b4846d3dbc7c4220ec145/firmware/src/BSP/stepper_controller.c#L430

Anyway, I am posting to report that I was able to run TLE5012B at 18 MHz (mbit).  More might be possible. The datasheet doesn't specify the maximum:
[image]