Infineon/TLE5012-Magnetic-Angle-Sensor #11: SPC configured sensors require a trigger pulse - need hardware reset

MorganSandercock, 2020-02-03:
If you accidentally purchased one of the E9xxx configured sensors and you're wondering why the angle always comes out as zero on the SSC (SPI) interface, then there's a note buried at the bottom of the Mode 4 register table on page 96 of the User's Manual which says...

"In SPC interface configuration, the sensor’s digital signal processing unit (DSPU) runs only when receiving a SPC trigger pulse on the IFA pin. This means that changes to register settings are applied and also the angle (AVAL) register is updated only after a trigger pulse"

So if you have one of those chips, you can't even change register settings to put it into a different mode that doesn't require a trigger unless you give it a trigger! The trigger pulse has some specific timing requirements, due to the clever way that the SPC mode can put 4 sensors on one data line.

There is a solution...  Initiate a hardware reset with the Activation Status Register then write a new value to the Mode 4 register within 120 microseconds of the reset. Then you can put it into PWM mode or any other mode that doesn't require a trigger. Put this code in your setup()...

```
  uint16_t mod4, data;
  Tle5012b.readIntMode4(mod4);
  mod4 = mod4 & 0xFFFD;  //sets PWM mode, doesn't affect any other mode settings
  Tle5012b.readActivationStatus(data);
  data = data | 0x0001;  //hardware reset bit (without affecting any other settings)
  Tle5012b.writeActivationStatus(data);
  delayMicroseconds(10);  //datasheet doesn't specifiy a delay but it does say we must write the register within 120us of the reset
  Tle5012b.writeIntMode4(mod4);
  //Now you are in PWM mode and out of SPC mode
```

OlafFilies, 2020-02-04:
Hello Morgan, thanks for this input. I didn't had the bulk E9xxxx for testing, only the ones on the evaluation boards (E1000 and E5000 typs). So I will try to get a E9000 type, test the code and implement it into the next release.

Ben6363, 2020-04-09:
Hello,

Recently, I bought the TLE5012 E9000. I thought that I might use it in a way of plug & play. That works when we talk about the TLx5012b 2go EvalKit application. But unfortunately it does not work when we talk about the Arduino IDE and the examples. But that it works on the IDE and that I can modify the code is actually what I would like to do.

So on the 1st try I just tried the Read_Test example. The result was that I only saw the angle of 360° and a temperature of 54°C which is not changing.
[image]

The result of the 2nd test was, that I inserted the code of @MorganSandercock , but unfortunately that did also not work out.
[image]

On the 3rd try I put the code of @MorganSandercock in another spot. But that did not work out as well.
[image]

Could anyone help me with that? That would be great. :)

MorganSandercock, 2020-04-09:
See that "check error"? Use it. Print it out. See what the error is. Add some more error-checking elsewhere, like make sure it returns the correct responses to my code.

Then check that it's not giving you a magnet-out-of-range error (shows as a system error.)

Does it return a valid value when you trigger it "manually" using the IFA pin?

OlafFilies, 2020-12-03:
To get the E9000 variant with the SPC interface running, read the TL5012B section of this interface carefully. The tricky thing is, that inside a 90 UT time frame the IFA pin has to go from high to low for at least 12 UT and than back. The meaning of UT is unit time which is a multiplication with the base time setting. This base time is default 3µs, so that means: we have to set IFA pin low for 12*3µs = 36µs. After that we need to wait for (90-12)*3µs = 224µs before we can start to read the data from the sensor. The IFA pin on the Sens2Go Kit has the pin number 9. You can also take the new example _E9000_SPC.ino_ from the develop branch. The develop branch will be the next release of the library based on a frame work setup which allows us to implement different MCUs with the same core library. It also includes a full set of register functions to read or set any part for the DSP registers.

OlafFilies, 2021-02-12:
The new release 3.0.0 includes the E9000SPC.ino example which describs the handling of the SPC predefined sensor. So I will close this issue. Thanks to all for the discussion and for point to this problem.

closed 2021-02-12 by OlafFilies