Infineon/TLE5012-Magnetic-Angle-Sensor #12: TLE5012B_E1000_MS2GO SPI Problem

janbaz497, 2020-12-03:
Hello,
I use the product TLE5012B_E1000_MS2GO and have a problem with the SSC.
I start the GUI and set SPI, but nothing happens there. The Auto Detected Log says:

SSC IF Not Active,
Active Interface: not known,
Key parameters: -not available.

How can I activate the SSC or where exactly is the problem?
I measure MOSI,MISO,CLK but no SPI Signal is there.

Here is the Eval Manual:

http://www.farnell.com/datasheets/2700284.pdf

The arduino examples dont works too.

Many thanks for the help!

janbaz497, 2020-12-03:
Here is a pic of the UI Auto Detected Log:
[image]

OlafFilies, 2020-12-03:
Hello janbaz497 thanks for asking. I think from what I see in picture you are simple not coinnected to the Sensor. I the left upper part you see the XMC Device: COM4. Klick on that and then on the left most icon below this box, witch is the connect. Than select the Sensor name and Interface, it should show values. If not than please check if COM4 is really the port where the sensor is connected to, check the cabling and check if there is maybe another piece of hardware on the same port so that they may overlapp. BTW, is the green LED on Sens2Go kit switched on?

janbaz497, 2020-12-05:
Hello Mr. Filies,

the board's LED lights up.
The incremental mode also works perfectly, so it cannot be due to the connection.

Can we also write in German?

OlafFilies, 2020-12-06:
Klar geht das auch in deutsch:-) . IIF funktioniert aber SSC nicht, mmh... alternativ vielicht mal das Read_Sensor_Type Beispiel mit der Arduino IDE ausprobieren.  Bzw. ist denn ein Magnet in der Nähe des Sensors? Ich werde das morgen mal mit einem IIF Sens2Go Kit ausprobieren ob diesen Fehler reproduzieren kann.

janbaz497, 2020-12-08:
Perfekt :)

Ich habe mal den Read_Sensor_Type Skript ausgeführt und der Serielle Monitor gibt folgendes aus:
[SensorType.txt](https://github.com/Infineon/TLE5012-Magnetic-Angle-Sensor/files/5658364/SensorType.txt)

In der GUI funktioniert tatsächlich nur der Inkrementalmodus.
Bei SPI bewegt sich nichts. Bei dem PWM und SPC Modus springt der Winkelanzeiger sinnlos hin und her.

Ich benutze folgenden Magneten https://www.neomagnete.de/de/detail/index/sArticle/212 .

OlafFilies, 2020-12-08:
Ok, ich habe die Ausgabe von Read_Sensor_Type gesehen, das kann nicht stimmen, oder anders gesagt, das SSC Interface ist tot. Checkerror = FF ist eine System Fehler und auch das fast alle Register nur FF zurückgeben ist so nicht richtig. Das aber IIF noch funktioniert liegt daran das der Sensor diese über die IFA/B/C pins ausließt. Auch die Tatsache das PWM und SPC keine vernünftigen Ergebnisse erzeugen liegt wohl daran das diese Interface nicht mehr über SSC programiert werden können. Fragen: Ist an dem Sensor gelötet worden?, sind evt. Kontakte kurzgeschlossen oder evt. gekappt? Mit einem Durchgangzprüfer mal alle Kontakte zwischen MCU und Sensor überprüfen. Die MCU splbst scheint zu funktionieren, denn es es gibt wohl keine Problem beim flaschen, weder mit der Arduino IDE noch mit der Evalkitsoftware und auch der Sensor selbst liefert noch Werte im IIF mode, also bleibt eigentlich nur noch ein Kontaktproblem oder die SSC Schnittstelle im Sensor ist tot.
Der Magnet ist Ok, es ging nur darum das ein Magnetfeld in der Nähe des Sensors ist.

janbaz497, 2020-12-08:
Gibt es einen Schaltplan/Layout wo ich die Kontakte nachvollziehen kann?
Dann würde ich das mal messen.
Ja ich habe direkt am Anfang Stiftleisten angelötet.
Kann keinen Kurzschluss erkennen.
Andere Frage: Falls ich das Teil geschrottet habe, kann man als Privatkunde das Evalboard auch in Deutschland bestellen?

OlafFilies, 2020-12-08:
Einen Schaltplan für das Evalboard habe ich jetzt nicht aber folgendes sollte zum messen der Kontakte reichen [pintout](https://github.com/Infineon/TLE5012-Magnetic-Angle-Sensor/wiki/Hardware-Platforms). Was den Sensor selbst angeht so gibt es hier eine vollständige Anleitung [Manual](https://www.infineon.com/dgdl/Infineon-Angle_Sensor_TLE5012B-UM-v01_02-en-UM-v01_02-EN.pdf?fileId=5546d46146d18cb40146ec2eeae4633b) und hier gibt es Links Distributoren die diesen Sensor verkaufen [Infineon Magnetic Sensor Kits](https://www.infineon.com/cms/en/product/promopages/sensors-2go/).

janbaz497, 2020-12-21:
Habe mir jetzt ein neues Evalboard besorgt.
Habe wohl das alte irgendwie abgeschossen.

Andere Frage...
Unter folgenden Link werden die Registerfunktionen genannt und unten steht ein Link der diese genauer beschreiben soll, leider funktioniert der Link nicht.
https://github.com/Infineon/TLE5012-Magnetic-Angle-Sensor/wiki/TLE5012B-REG-API

Können sie mir da helfen?

OlafFilies, 2021-01-11:
> https://github.com/Infineon/TLE5012-Magnetic-Angle-Sensor/wiki/TLE5012B-REG-API

Hallo, sorry bin heute erst wieder im Büro... Das wiki bezieht sich auf die neue Version der TLE5012 lib. Die ist aber noch nicht released, weshalb auch Teile der doku noch nicht komplett sind. Die Lib ansich ist aber schon benutzbar und befindedt sich im develop branch.

closed 2021-07-12 by OlafFilies