Infineon/TLE5012-Magnetic-Angle-Sensor #15: Tle5012b::getAngleSpeed heap overflow?

TazerReloaded, 2021-04-12:
**Describe the bug**
We are working on a rewrite of the firmware for a S42B closed loop stepper which uses the TLE5012 as feedback. Hardware details [here](https://www.biqu.equipment/de/products/bigtreetech-s42b-v1-0-closed-loop-driver-control-board-42-stepper-motor-oled-3d-printer-parts).
Because of the hardware design we cannot use the official library, but we are using it as reference. Reading and writing registers works, we can get valid angle values from the sensor, but I can't figure out how the angle speed calculation is supposed to work.

[In line 486](https://github.com/Infineon/TLE5012-Magnetic-Angle-Sensor/blob/77b64e2817fe84fc2df79394474dc7932383ff0f/src/corelib/TLE5012b.cpp#L486) an array with 5 slots is defined. [A bit further down](https://github.com/Infineon/TLE5012-Magnetic-Angle-Sensor/blob/77b64e2817fe84fc2df79394474dc7932383ff0f/src/corelib/TLE5012b.cpp#L509), values from array location 6 are read. This could be anything that happens to be in memory at that location, or am I missing something?
Furthermore, the comments are not very helpful trying to figure out what the different factors are doing. What is `firMDVal` supposed to do? `intMode2Prediction` seems to determine if angle speed prediction is enabled or not, and `rawAngleRange` specifies the possible values of the raw angle?
I'm trying to find the correct register locations to get this function working, currently the output makes absolutely no sense.

**Expected behavior**
Function should return angle speed, but returns nan/inf.

**Software (please complete the following information):**
 - OS: Windows 10
 - IDE name and version: Visual Studio Code 1.55.0

**Hardware (please complete the following information):**
 - Sensor with full name: TLE5012B-E1000
 - Shield with full name : Custom hardware, sensor installed opposing a magnet on the end of a stepper motor shaft.

**Additional context**
Our repository: https://github.com/CAP1Sup/Intellistep

9Volts9er, 2021-04-13:
That's true, as I see it, it should be `int8_t numOfData = 0x6;` in [line 486](https://github.com/Infineon/TLE5012-Magnetic-Angle-Sensor/blob/77b64e2817fe84fc2df79394474dc7932383ff0f/src/corelib/TLE5012b.cpp#L486).

firMDVal, intMode2Prediction and rawAngleRange represent sensor settings that affect the speed calculation formula.
Therefore the code should read the angle speed register 0x02 + 6 additional registers to have the registers 0x06 and 0x08 included in the sensor respond.

The value ANG_SPD is basically just the difference between the current value and the second-to-last measured angle value. So you need to know how much time has passed between the two values to caluclate the correct speed. This is given by the FIR_MD setting (bits 15-14) in register 0x06, which controls the updateRate `t_upd` of the sensor. 

The next thing you need to know for the speed calculation is whether the sensor operates in prediction mode or not. This is given by bit 2 in register 0x08. According to the user manual you either have to use in the formula `2 * t_upd` when prediction is disabled or `3 * t_upd` when it is enabled.

The last important setting for this calculation is the sensors angle range. The sensor can be configured to either measure the full 360° or just 180° or even less. For your application this setting probably doesn't make sense, so you could always use the full 360° range in the formula.
Also, if you know your sensor settings and they cannot be changed in your application, you may want to use a constant calculation formula. Then you do not need to readout the 6 additional registers, but just the 0x02 register.

You can find the information also at following pages in the [User Manual](https://www.infineon.com/dgdl/Infineon-Angle_Sensor_TLE5012B-UM-v01_02-en-UM-v01_02-EN.pdf?fileId=5546d46146d18cb40146ec2eeae4633b):

- page 82: description of angle speed register (0x02) with the calculation formula
- page 85: description of register 0x06 with the update rate settings (FIR_MD)
- page 88: description of register 0x08 with angle range and prediction mode settings

I hope, this helps.

TazerReloaded, 2021-04-13:
Thank you for the quick and detailed response!
I changed line 486 to 0x06 and now, with your explanation, the register values make sense and the calculation seems to be correct.

djaumann, 2021-04-13:
@TazerReloaded Thanks for the feedback. I will close the issue then.
Feel free to raise another issue in case of other problems

closed 2021-04-13 by djaumann

IhorNehrutsa, 2021-07-12:
@djaumann You closed the issue, but bug is still in code. 
Please, Merge PR: Fixed the bug in the getAngleSpeed function #16 to resolve the issue.

djaumann, 2021-07-12:
@IhorNehrutsa Thank you for mentioning this! I just merged the changes to the master. Everything should be up to date now.
Let me know if there are more problems.

And sorry for the inconveniences ...