Infineon/TLE5012-Magnetic-Angle-Sensor #17: esp32 support

SuCheng04, 2021-06-11:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Is your solution generic or only for a fixed setup**
We prefer generic solutions which work on any kind of Hardware/Software combination in range of this software.
So please try to not generate a feature for only a very limited setup.

**Additional context**
Add any other context or screenshots about the feature request here.

jaenrig-ifx, 2021-06-14:
Hi @SuCheng04,

I can see you have not completed the template. Could you describe your issue or request?

SuCheng04, 2021-06-15:
Sorry, yea, so when I run the E9000SPC example, I get an error where all the pins in the TLE5012-pal-ino.hpp were not declared. How would I fix this?

SuCheng04, 2021-06-16:
Ok, so I was able to solve my previous problem, but now, I get an issue where "'en' was not declared in the scope" (referring to src/corelib/TLE5012b.hpp line 117,125,137,144)

jaenrig-ifx, 2021-06-17:
Hi @SuCheng04,

Which board are you using? Or Arduino core...

[image]

SuCheng04, 2021-06-17:
I am using the ESP32 Pico Kit

OlafFilies, 2023-10-23:
Hello
for ESP32 support you can use my fork here https://github.com/OlafFilies/magnetic-angle-sensor. The changes there will go into this repo as soon as a new version will be released here.

ederjc, 2025-03-10:
This library has been replaced by the new [XENSIV™ TLx5012B Magnetic Angle Sensor Arduino Library](https://github.com/Infineon/arduino-xensiv-angle-sensor-tlx5012). If the issue persists, please re-open it in the new library repository.

closed 2025-03-10 by ederjc