Infineon/TLE5012-Magnetic-Angle-Sensor #19: PIN_SPI_SS is not defined for some board distributions

CreativeRobotics, 2022-02-14:
This won't compile for any Adafruit SAMD board because PIN_SPI_SS is not defined in this board distribution:

In file included from d:\dropbox\arduino\libraries\tle5012b\src\framework\arduino\pal\tle5012-pal-ino.hpp:28,
                 from D:\Dropbox\Arduino\libraries\TLE5012B\src/framework/arduino/wrapper/TLE5012-platf-ino.hpp:31,
                 from D:\Dropbox\Arduino\libraries\TLE5012B\src/TLE5012-ino.hpp:27,
                 from D:\Dropbox\Arduino\libraries\TLE5012B\examples\readAngleTest\readAngleTest.ino:18:
d:\dropbox\arduino\libraries\tle5012b\src\framework\arduino\pal\spic-arduino.hpp:44:28: error: 'PIN_SPI_SS' was not declared in this scope; did you mean 'PIN_SPI_SCK'?
   44 |      SPICIno(uint8_t csPin=PIN_SPI_SS);

CreativeRobotics, 2022-02-14:
Fixing this still won't make it work with a lot of Arduino distributions because the SPI class isn't handled correctly:

> D:\Dropbox\Arduino\libraries\TLE5012B\src\framework\arduino\pal\spi3w-ard.cpp: In constructor 'SPIClass3W::SPIClass3W()':
> D:\Dropbox\Arduino\libraries\TLE5012B\src\framework\arduino\pal\spi3w-ard.cpp:27:35: error: no matching function for call to 'SPIClass::SPIClass()'
>    27 | SPIClass3W::SPIClass3W():SPIClass()
>       |                                   ^
> In file included from D:\Dropbox\Arduino\libraries\TLE5012B\src\framework\arduino\pal\spi3w-ino.hpp:14,
>                  from D:\Dropbox\Arduino\libraries\TLE5012B\src\framework\arduino\pal\spi3w-ard.cpp:11:
> C:\Users\bbigg\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.9\libraries\SPI/SPI.h:115:3: note: candidate: 'SPIClass::SPIClass(SERCOM*, uint8_t, uint8_t, uint8_t, SercomSpiTXPad, SercomRXPad)'
>   115 |   SPIClass(SERCOM *p_sercom, uint8_t uc_pinMISO, uint8_t uc_pinSCK, uint8_t uc_pinMOSI, SercomSpiTXPad, SercomRXPad);
>       |   ^~~~~~~~
> C:\Users\bbigg\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.9\libraries\SPI/SPI.h:115:3: note:   candidate expects 6 arguments, 0 provided
> C:\Users\bbigg\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.9\libraries\SPI/SPI.h:113:7: note: candidate: 'constexpr SPIClass::SPIClass(const SPIClass&)'
>   113 | class SPIClass {
>       |       ^~~~~~~~
> C:\Users\bbigg\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.9\libraries\SPI/SPI.h:113:7: note:   candidate expects 1 argument, 0 provided
> C:\Users\bbigg\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.9\libraries\SPI/SPI.h:113:7: note: candidate: 'constexpr SPIClass::SPIClass(SPIClass&&)'
> C:\Users\bbigg\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.9\libraries\SPI/SPI.h:113:7: note:   candidate expects 1 argument, 0 provided
>

OlafFilies, 2023-10-23:
Hello, please refer to this fork https://github.com/OlafFilies/magnetic-angle-sensor, it has a definition for the PIN_XXX macros if they are missing. The changes of the fork will be also ported into the next release.

ederjc, 2025-03-10:
This library has been replaced by the new [XENSIV™ TLx5012B Magnetic Angle Sensor Arduino Library](https://github.com/Infineon/arduino-xensiv-angle-sensor-tlx5012). If the issue persists, please re-open it in the new library repository.

closed 2025-03-10 by ederjc