Infineon/TLE5012-Magnetic-Angle-Sensor #21: Python wrapper

lidar08, 2024-02-12:
I'm looking for a simple solution for integrating TLE5012 into my python project. 
A simple and clean solution would be to use your code with a python wrapper.

OlafFilies, 2024-02-12:
Hello, thanks for your question. We currently are working on a new release of the library this will manly fix some bugs and it will get a new setup which allows us to implement different echo systems. This will yet not include Python but maybe a microPython port if this what you are searching for. BTW there is some example code in Processing which interacts with the data output via a serial interface, maybe this also helps.

ederjc, 2024-02-14:
Hi @lidar08,
Your idea of building a Python wrapper on top of the library is feasible. We did this for other libraries before. You can use e.g. [this project](https://github.com/Infineon/multi-half-bridge) as reference. From our side there is currently no plan to extend this library by a Python wrapper.
Best regards
Julian

closed 2024-02-14 by ederjc