Infineon/TLE5012-Magnetic-Angle-Sensor #22: RP2040 Compatibility

orlando-helfer, 2024-04-03:
Hello, 
are there any plans to add compatibility with the family of boards based on the RP2040 (with the arduino-pico core)? 
I am using a custom board with a RP2040 and have the Sensor (variant TLE5012BE1000XUMA1) connected to the SPI bus of the MCU. A 510 Ohm Resistor between MOSI and MISO is also in place. 
Unfortunately the compilation using the Arduino Library is not functional for this chip atm. and I'm having some trouble to implement it for the this chip :/

OlafFilies, 2024-04-03:
Hello,
and thanks for this question.
Yet at the moment we do not plan to convert the code to RP2040. but the TLx5012 code is yet under review and will be updated soon.  This will not solve directly your problem but there is a good chance that we implement a MiroPython port of the library. If this is not what you need, than maybe the new Arduino lib will help as it is more clear and seperates the core functions better from the Arduino function, as you only need the wrap the SPI and GPIO functionality arround the core lib.

OlafFilies, 2024-04-03:
Hello,
there is yet no final release date and it still will take some time and it will be certainly after embedded 2024 trade fair. The only thing in between which I can offer is my fork of the repo which includes almost all changes so far (https://github.com/OlafFilies/xensiv-angle-sensor-tlx5012).

orlando-helfer, 2024-04-03:
Alright, I'll take a look!
Thanks'

ederjc, 2025-03-10:
This library has been replaced by the new [XENSIV™ TLx5012B Magnetic Angle Sensor Arduino Library](https://github.com/Infineon/arduino-xensiv-angle-sensor-tlx5012). If the issue persists, please re-open it in the new library repository.

closed 2025-03-10 by ederjc