Infineon/TLE5012-Magnetic-Angle-Sensor #23: TLE5012B_E1000_MS2GO Based Custom Design Schematic Help

walnut-engineering, 2024-04-03:
**Describe the bug**
Sorry wasn't sure where else to post this.  

The code works fine on [TLE5012B_E1000_MS2GO](https://www.infineon.com/cms/en/product/evaluation-boards/tle5012b_e1000_ms2go/) using the XMC1100_VQFN24 but on custom design using XMC1100_T016 I am unable to get communication with the TLE5012B sensor, I always get a response of SYSTEM_ERROR when trying to initialize the sensor.  I am hoping there is just something I am missing on the difference between the devices, but I not seeing what the issue is.  I see there are some GPIO used in addition to the SPI, but it didnt seem to apply or would interfere with functionality.  I believe the schematic is correct, but its difficult to compare as datasheet schematic image is of poor quality and source PCB files are not available for the dev kit.  I have the PCB files and clear schematic for the [XMC 2Go](https://www.infineon.com/cms/en/product/evaluation-boards/kit_xmc_2go_xmc1100_v1/) which is very similar just does have the connection to TLE5012.   

**To Reproduce**
Run code on TLE5012B_E1000_MS2GO with XMC1100_VQFN24, TLE5012.begin returns NO_ERROR
Run code on TLE5012B_E1000_MS2GO based custom design with XMC1100_T016, TLE5012.begin returns: SYSTEM_ERROR

**Expected behavior**
Code runs as expected using TLE5012B_E1000_MS2GO but when 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Software:**
 - OS: Windows
 - IDE: VSCode
 - Libraries with full name and version: infineon/TLE5012@^3.1.0
 - XMC-for-Arduino with version: Infineon/XMC-for-Arduino/releases/tag/V3.0.0

**Hardware:**
 - Sensor with full name: TLE5012BE3005
 - Shield with full name : TLE5012B_E1000_MS2GO & TLE5012B_E1000_MS2GO Based Custom Design
 [image]
 - Circuit diagram: 
[image]

OlafFilies, 2024-04-04:
Hello, nice board :-). Can you try to put a resistor between P0_6 and P0_7 like mentioned here https://github.com/Infineon/TLE5012-Magnetic-Angle-Sensor/wiki/Hardware-Platforms. One other point is, that the library will soon get a major upgrade, but is yet not released. You can get a beta from my fork but that is the full setup which does not directly compile under Arduino. But I would try the resistor first as I have seen some of these problems with on different MCUs.
If the code can help, I can make an Arduino lib out of it, but this will need some time as I'm next week on the embedded 2024 trade fair. The code can be found here https://github.com/OlafFilies/xensiv-angle-sensor-tlx5012

walnut-engineering, 2024-04-04:
Thanks! 

Ohh I had missed that, Ill give it a try, thanks for the help!

walnut-engineering, 2024-04-06:
@OlafFilies adding resistor between P0_6 and P0_7 does make it work, ill update my schematic, thanks again!

closed 2024-04-06 by walnut-engineering

OlafFilies, 2024-04-08:
@walnut-engineering your welcome, btw., cool design, what is the purpose of this board.

walnut-engineering, 2024-04-10:
@OlafFilies 😁 The design is meant to configure the pole pair setting of the E3005 hall switch mode on each startup.  the default setting of 5 pole pair, is not suitable for the motor and controller I am using. Hopefully I didn't miss something and there is a way to 'write' that config such that it persists power cycles.  I plan on uploading the full design to my GitHub when complete.  While I am at it I plan on adding some feedback based on rotation, like flash the led on each revolution..

OlafFilies, 2024-04-16:
Hello, nice setup :-). The TLE5012 is normally preset for the interface but set diodes not include special register settings. To change the number of pole paris you must set the HSM_PLP  field in the MOD_4 register (page 94 ff in the manual). This setting has to be done again after a reset or a power off, but it can be part of the initial process startet by the XMC which is also on the board, so I would say that this works. Also keep in mind that the revolutions counter (register REVOL/AREV) is a 9 bit signed integer. So if you it overflows after 255 revolutions.