Infineon/TLE5012-Magnetic-Angle-Sensor #25: readBlockCRC accesses beyond array bounds

maxschommer, 2024-08-03:
**Describe the bug**
It seems that _registers[9] is being accessed in this function despite the array having length 9. I'm not exactly sure what the intent of that was, since only 9 bytes are being sent so the last byte isn't being used.

```
// begin read functions
errorTypes Tle5012b::readBlockCRC()
{
	_command[0] = READ_BLOCK_CRC;
	_registers[CRC_NUM_REGISTERS + 1] = {0};  // Number of CRC Registers + 1 Register for Safety word
	sBus->sendReceive(_command, 1, _registers, CRC_NUM_REGISTERS+1);
	errorTypes checkError = checkSafety(_registers[8], READ_BLOCK_CRC, _registers, CRC_NUM_REGISTERS);
	resetSafety();
	return (checkError);
}
```

OlafFilies, 2024-08-05:
Hello, you are right that CRC_NUM_REGISTERS has only a length of 0x0008, which are the registers from address 08 to 0F, but has the TLE5012 is a an automotive sensor, every communication includes an additional safety word. So it is not the number of registers, it is the sendReceive command which needs the additional size for the safety word. The function checkSafety than runs a CRC check between the content (8 registers) and the safety word (register 9). 
You will find this type of check in all parts of the TLE5012 core lib as it is a functional safety (ISO26269) product.
It is possible to switch of the safety word in some functions, but not for this one here.
So this is not a bug, it is a feature of this type of sensor.

maxschommer, 2024-08-05:
Hi Olaf,

I understand that an additional safety word causes the _registers length to be 9 (8 CRC + 1 safety). However, accessing `_registers[CRC_NUM_REGISTERS + 1]` accesses the **10th element**, which seems like a mistake. Perhaps I am still misunderstanding, apologies.

OlafFilies, 2024-08-06:
Hello, sorry, but I don't see a 10th element and I checked with the JLink debugger. The array _registers is of size [9] and it runs from 0 to 8 with the safetyWord at _register[8]. please have a look at the attached picture.
[image]

maxschommer, 2024-08-06:
So line 320 in the picture is what I'm talking about. I might be misunderstanding how assigning a `{0}` works, but the access of `_registers[CRC_NUM_REGISTERS + 1]` seems out of bounds. From what I see,

`_registers[CRC_NUM_REGISTERS + 1]` -> `_registers[8 + 1]` -> `_registers[9]` == 10th element

ederjc, 2025-03-10:
This library has been replaced by the new [XENSIV™ TLx5012B Magnetic Angle Sensor Arduino Library](https://github.com/Infineon/arduino-xensiv-angle-sensor-tlx5012). If the issue persists, please re-open it in the new library repository.

closed 2025-03-10 by ederjc