Infineon/TLE5012-Magnetic-Angle-Sensor #26: Trouble connecting to Arduino Nano

jyl087, 2024-09-08:
**Describe the bug**
I'm trying to interface the TLE5012B E1000 to an Arduino Nano and am having a hard time making it work reliably. The sensor is wired to the Nano D11,D12 (data), D13 (sck), and D10(ss).  I am able to compile and run the sample "readAngleTest.ino" program and it seems to work reliably.

However, I am trying to change the code so that every 50mS I want the Arduino to just print the angle and the temperature: nothing else. But when I reduce the amount of output that gets printed, it seems like the program freezes and the data that I get is all zeros.  
`
  Serial.print("Temperature:"); Serial.print(t);
  Serial.print("°C\tangle:");   Serial.print(d);
  Serial.print("°\trange:");    Serial.print(r);
  Serial.print("\tspeed:");	    Serial.print(s);
  Serial.print("\t Revolution:");Serial.println(b);
`
If you get rid of all the English words and just print the numbers with commas as separators, the program stops working! There seems to be some interaction between the USB Serial line and the SPI interface. Or maybe some other explanation for why this is happening?
And even more strange, if I simply print more garbage characters, like this:
`
  Serial.print("The quick brown fox jumps over the lazy dog");
`
It starts working again! It's really the strangest thing that the amount of characters printed to the console is affecting the TLE chip. 

One more thing is that sometimes if I unplug the whole board and reboot it, it will start working again. Is there some way to reset the TLE5012 from the Arduino?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the readAngleTest.ino file.
2. Edit the Serial.print in the loop to print just the angle and temperature and nothing else.

**Expected behavior**
The requested data to be printed to the USB console

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Software (please complete the following information):**
 - OS: Ubuntu 22.04
 - IDE name and version: Arduino 2.3.2
 - Libraries with full name and version: TLE5012B 3.1.0
 - XMC-for-Arduino with version: not sure

**Hardware (please complete the following information):**
 - Sensor with full name: TLE5012 E1000
 - Shield with full name :  [TLE5012](https://www.amazon.com/dp/B0C4ZR57G9/)
 - Circuit diagram if needed: 
   - Shield P1.3 - D11,D12
   - Shield P1.4 - D10
   - Shield P1.5 - D13

**Additional context**
None

ederjc, 2024-09-09:
Hi @jyl087,

I did the following steps and tried to reproduce your issue:
1. load `readAngleTest.ino` example
2. Adapt the code by removing all print statements except: `Serial.print(t); Serial.print(","); Serial.println(d);`
3. Changed the delay to 50ms

I could unfortunately not reproduce your error. For me it runs without interruption. I am using [TLE5012B E1000 Kit 2Go](https://www.infineon.com/cms/en/product/evaluation-boards/tle5012b_e1000_ms2go/), so maybe the issue is related to the Nano clone. Can you test with another controller?

OlafFilies, 2024-09-09:
Hello, question, so you are using the bare chip with an Arduino Nano or a brake out board or is this a different setup? Or is it a XMC2Go kit where you simply wire the pins to an Arduino Nano?
A bare chip and a brake out board may need some additional resistors. We are also preparing a new version of the library right now but I have to make an Arduino port from that. I will give you a note as soon as this port is ready. 
BTW, sometimes I get also sometimes problems on Arduino clones which are not appearing on the original hardware, so they are maybe not really 100% identical which makes it hard to track.

jyl087, 2024-09-09:
Hello all. I appreciate your quick replies! As noted in the original ticket, the breakout board is this: 
https://www.amazon.com/dp/B0C4ZR57G9/
It does include resistors.

The Nano clone is this:
https://www.amazon.com/gp/product/B07G99NNXL
I did try to swap out the Nano with an identical unit (they came in a 3 pack), but same behavior. Not sure how such a simple board could go wrong, but I have some ESP32s laying around. I don't think the library is compatible though?

ederjc, 2024-09-09:
Hi @jyl087,

> I have some ESP32s laying around. I don't think the library is compatible though?

@OlafFilies will get back to you regarding ESP32.

> When you said you tried to reproduce, it looks like you weren't using a Nano but an Infineon eval board, right?

Correct, I tested with an Arduino-compatible Infineon eval board.

> So it could be a problem with Nano, clone or not, right? I.e. a "genuine Nano" wouldn't make things better?

hmm, depends:
1. How do you flash these boards? Did you have to install it through Arduino's board manager or is it natively supported by just selecting *Arduino Nano*?
2. Have you tried running sample code without our library where you output e.g. a counter with a high frequency (delay = 50ms)? Does this work without any issues?

OlafFilies, 2024-09-09:
Hello, you can try this very new port of the [TLx5012B repo](https://github.com/Infineon/arduino-xensiv-tlx5012b-angle-sensor) which includes many bug fixes and an EPS32 port. As this is still code in progress and review it may not work but you can give it a try.

jyl087, 2024-09-09:
@OlafFilies - THANK YOU! It seems to work. No more issues with the console interacting with the SPI interface. 
I will definitely try out the ESP32 support as well. Just hadn't gotten around to playing with those yet, but clearly a much more capable board than the Nano!
Thanks again.

jyl087, 2024-09-09:
@OlafFilies , Since you mentioned the ESP32, I decided to give it a try. I have this ESP-WROOM-32 dev board: https://www.amazon.com/gp/product/B0CGQV7P9K/
I verified the SPI pin assignments by printing them in the setup() phase:
MISO: 19
MOSI: 23
SCK: 18
SS: 5

And that is how I wired up the TLE5012B. Further, I took power and ground from the Vin / Gnd on the ESP32 board. I verified that there was 4.7V on the Vin pin. However, I'm getting a checkError: 1 during startup.

Please let me know if there's anything you'd like me to try to get this library working with the ESP32. Thanks!

OlafFilies, 2024-09-10:
Hello, for a quick answer, as the TLx5012Bs all use 3wire, you have to connect MISO/MOSI together and as far as I remember you need a resistor between them which is bigger than the usual 470 Ohm. I think it was 1 MOhm .
Sorry for that, but as I wrote, this is still a work in progress repo without proper documentation.

jyl087, 2024-09-10:
Hello @OlafFilies I tried a quick test with the 1MΩ resistor on the data line and unfortunately it still says checkError = 1 during setup. I verified that the TLE is getting 4.7V (not quite 5) but I think that should be ok.

At least for now, the Nano seems to be working reliably, so I'll stick with that. I'm happy to test the ESP32 if you have any new software to try in the future.

OlafFilies, 2024-09-11:
Hello, thanks I appreciate that you test this. May have a look into the [manual](https://www.infineon.com/dgdl/Infineon-Angle_Sensor_TLE5012B-UM-v01_02-en-UM-v01_02-EN.pdf?fileId=5546d46146d18cb40146ec2eeae4633b) of the TLE5012. There you can find under 3.7 and 3.8 two pictures how the sensor can be connected, it is the SSC push-pull or SSC open-drain configuration and as far as I remember the ESP only works with the open-drain config. That means two 100 Ohm resistors between DATA and MISO/MOSI and a 1kOhm resistor to GND. I think you can leave the 100 Ohm resistors out and only use the 1kOhm (maybe it was only 1kOhm not 1MOhm). Anyway it was a little more tricky than for other MCUs which work in the push-pull configuration. If this is not working, I have to test how that and of course I have to rewrite the documentation for that which will be done in the next weeks.

jyl087, 2024-09-11:
hello @OlafFilies , unfortunately, the breakout board I'm using for the sensor already has a 470 Ω soldered in place and I don't want to modify it. I did, however, try to hook it up in the open drain mode using a 4.7k pullup and a 470Ω on the MISO/MOSI pins as well just to see if that would work. I was thinking to try to maintain the 1:10 pulldown ratio.

I downloaded the sample angle reading code, but unfortunately it still gives a checkError 1. 

By the way, I couldn't find it in the documentation, but is there a recommended airgap between the sensor and the magnet? The kit I bought doesn't really specify anything about the strength of the magnet, but it is 6mm in diameter.

OlafFilies, 2024-09-11:
Hello, yes there is a recommended gap, but the sensor is very insensitive against that, and also quiet robust, Switching cables during runtime does not really matter, also inserting or removing resistors are normally not a problem.
BTW, it is not a problem with the gap but more of the strength of the magnetic field.  There has to one for the initial check and it should be in a reasonable range. Also it works with the Nano....

You try the PCB you mentioned here https://www.amazon.com/dp/B0C4ZR57G9/ ? 
Anyway, it will take a little bit of time for me to track that down. The esp32 port for Arduino is not very good in 3wire SPI, which was the reason to use the open-drain setup.
The library should find the EPS32 automatically (see here spi3w-ino.hpp) which than runs the spi3w-esp32.cpp.

jyl087, 2024-09-18:
Hello @OlafFilies, I've started getting [1] error code during boot, which according to the documentation is a SYSTEM_ERROR. I've checked the voltage to the sensor: 5.03V. Could the sensor have somehow gone bad? Is there anything else I can check? Note, this is after the chip was working with no other changes.

OlafFilies, 2024-09-18:
Hello, you run the Sensor with 5V !? it should run only with 3.3V. It can withstand 5V, but it gets hot and yes you can destroy it. Give the Sensor one hour and than restart with 3.3V.

jyl087, 2024-09-18:
Thanks for the reply! I wasn't aware about the 3.3V. I will see about changing that.
However, I discovered that the problem was a broken data wire. It's back to working now.

jyl087, 2024-09-18:
I just checked the datasheet, and in Table 4, it says that the Supply Voltage should be between 3 and 5.5 with typ = 5.0. Is this information incorrect?

OlafFilies, 2024-09-20:
Hello, running the sensor on 3.3V is better for the life time and temperature drift. These sensor are rather stable you hotplug them, remove the magnet, using to strong magnets, use the onchip ADCs for other date and many more fancy this, but all of that will affect the sensitivity and life time, so better avoid that. Therefor if you can run with 3.3V than prefer that instead of using 5V, also it will run on 5V.

jyl087, 2024-09-20:
Ok, good to know! I haven't checked the Arduino specs but I assume that the SPI lines can work with 3.3v logic as well?

koombot, 2024-11-09:
Hi, I've been doing some setup with the same breakout board as the same commenters in the thread and have encountered similar issues with an arduino nano clone.
Could I just check if you have the magnet in front of the sensor when you power on the microcontroller and sensor?  I had the checkError: 1 perpetually until I made sure the magnet was in front of the sensor.  I've checked with several of the examples as this appears to be a typical behaviour.

OlafFilies, 2024-11-11:
Hello, so at the beginning of each code example the checkerror from the lib initial routine is printed out. This will mark if there was an error during startup of the sensor device. A missing magnetic field is one of these errors but not the only possible. If the sensor device is working, than you can normally ignore that, so attaching a magnet somewhat later will just lead to a working sensor without any further settings. This typically happens when you ask for new values in loop that it starts to printing values as soon as the sensor recognizes the magnetic field. 
Question, which type of Arduino Nano are you using?

ederjc, 2025-03-10:
This library has been replaced by the new [XENSIV™ TLx5012B Magnetic Angle Sensor Arduino Library](https://github.com/Infineon/arduino-xensiv-angle-sensor-tlx5012). If the issue persists, please re-open it in the new library repository.

closed 2025-03-10 by ederjc