Infineon/TLE5012-Magnetic-Angle-Sensor #28: Unable to install library with Arduino 2.3.3

koombot, 2024-11-08:
**Describe the bug**
I am unable to install the library through the Arduino library manager when using Arduino 2.3.3 as it says it is missing dependancies for xmc-for-arduino.  I have installed the xmc microcontroller board controller software and the segger j-link software.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Arduino 2.3.3 ide
2. Follow instructions to install xmc board controller, segger j-link outline here on installation instructions
3. Attempt to install tle5012 from Arduino library manager
4. See error about missing dependancies 

**Expected behavior**
I expected the library to be installed.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Software (please complete the following information):**
 - OS: windows 10
 - IDE name and version: Arduino 5.3.3
 - Libraries with full name and version: tle5012 version 3.1.0
 - XMC-for-Arduino with version: XMC-for-Arduino V3.4.1

**Hardware (please complete the following information):**
 - not applicable yet as can't install the library.

**Additional context**
Add any other context about the problem here.

OlafFilies, 2024-11-08:
Hello, and sorry for the inconvenience. Please take the this library [arduino-xensiv-tlx5012b-angle-sensor](https://github.com/Infineon/arduino-xensiv-tlx5012b-angle-sensor)) from the Infineon GitHub, the other one is outdated and will be archived soon.

koombot, 2024-11-08:
Okay.  Thank you.  That seems to have installed.  I installed as a zip.

closed 2024-11-08 by koombot